SouthEugeneRoboticsTeam/ursa #10: Generate SSID dynamically

andrewda, 2019-06-27:
https://github.com/SouthEugeneRoboticsTeam/ursa/blob/e544ecc3af908fd6746f0c8b8f17133ef61356e3/ursa.ino#L8

This line could instead read something like:

```cpp
const char *robotSSID = "SERT_URSA_" + std::to_string(ID);
```

Note: you'll have to `#include <string>` and probably move these lines down below the `#include`s.

joshua255, 2019-06-28:
Unfortunately, arduino doesn't have the standard string library. #include <string> gives the error "No such file or directory" Any other ideas? this issue is turning out to be harder than I expected.

andrewda, 2019-06-28:
@joshua255 This maybe?

```c
char robotSSID[20];
sprintf(robotSSID, "SERT_URSA_%d", ID);
```

joshua255, 2019-06-28:
I'll test that.
this code makes a string of what we want, but the wifi command needs a char pointer
```#define ROBOT_ID 0
String robotSSID = "SERT_URSA_";
void setup() {
  robotSSID += ROBOT_ID;
  Serial.begin(2000000);
}
void loop() {
  Serial.println(robotSSID);
  delay(1000);
}```

andrewda, 2019-06-28:
Here, updated to left-pad single digit numbers:

```c
char robotSSID[12];
sprintf(robotSSID, "SERT_URSA_%02d", ID);
```

joshua255, 2019-06-28:
That compiles for esp32 and works on an arduino micro. I'll make a pull request soon.

joshua255, 2019-07-16:
@andrewda Can you write up an explanation of how that function works for the URSA wiki? It looks like magic to me.

andrewda, 2019-07-17:
@joshua255 the sprintf function?

joshua255, 2019-07-17:
Yes, I don't know how it works.
Also what's the %

andrewda, 2019-07-17:
Yep I can write up a description. I'll write something more descriptive when I get back to my laptop, but `%02d` is essentially a placeholder that will be replaced by the following number, padded to a length of 2 with 0's left padding it. I believe the signature of `sprintf` is `sprintf(output, template, ...values)`

andrewda, 2019-07-17:
Here's a pretty in-depth description https://www.tutorialspoint.com/c_standard_library/c_function_sprintf

andrewda, 2019-07-19:
@joshua255 Okay sorry for the delay -- here's an in-depth explanation.

To generate the robot's WiFi network name, the code takes in a template (`SERT_URSA_%02d`) and an unique ID number for the robot. The `sprintf` function takes in 2 required parameters, with additional optional parameters that can be added to substitute into the `template`, giving it a signature of `sprintf(output, template, ...values)`. Substitutions in the template string are denoted by `%`s, followed by a `d`, `s`, or some combination of other options which can be found [here](https://www.tutorialspoint.com/c_standard_library/c_function_sprintf).

In this case, the `%02d` denotes a substitution of type integer (given by the `d`), left padded by `0`s to make it `2` characters long. For example, our function could be used like so:

```c
int ID = 5;
char output[12];
sprintf(output, "SERT_URSA_%02d", ID);
// output => SERT_USRA_05

ID = 93;
sprintf(output, "SERT_URSA_%02d", ID);
// output => SERT_USRA_93
```

andrewda, 2019-07-19:
Try it out here: https://onlinegdb.com/Sy-hih0Wr

Also note if `ID` is ever > 99, you'll get an exception

joshua255, 2019-07-19:
I added your explanation to the wiki. Thank you!

closed 2019-07-19 by joshua255