SouthEugeneRoboticsTeam/ursa #22: Find a better way to delay code

andrewda, 2019-06-28:
https://github.com/SouthEugeneRoboticsTeam/ursa/blob/8cda4d06551d58af749c5942b2c0c170dd76e7fb/ursa.ino#L74

I'm sure there's a better way than this, even if it's literally just a for loop. Something like this would probably suffice (note: completely untested):

```c
void delay(int timeNs)
{
  int cycles = timeNs * 0.24;
  for(int i = 0; i < cycles; i++) NOP();
}
```

joshua255, 2019-06-28:
Doesn't a for loop increment a counter each time, so it takes more than just the nop's clock cycle? @circuitbird

andrewda, 2019-06-28:
How precisely do you need to wait for 300ns...? Incrementing a counter shouldn't take long at all.

joshua255, 2019-06-28:
It's okay to wait bit longer, the requirement is that it is longer than 200ns. Doesn't a nop take a single clock cycle? Incrementing a counter couldn't take less time than one cpu instruction? This delay is in an interrupt, so timers might not work.

andrewda, 2019-06-28:
Here's something else I found online:

```c
void IRAM_ATTR delayMicroseconds(uint32_t us)
{
    uint32_t m = micros();
    if (us) {
        uint32_t e = (m + us);
        if (m > e){ //overflow
            while(micros() > e){
                NOP();
            }
        }
        while(micros() < e){
            NOP();
        }
    }
}
```

Obviously it only has microsecond granularity instead of nanosecond, but in the end I think the timing for this doesn't need to be _thaaaat_ precise.

joshua255, 2019-06-28:
inside an ISR, micros might act unreliably https://arduino.stackexchange.com/questions/22212/using-millis-and-micros-inside-an-interrupt-routine 
72 NOPs looks ugly, but I don't think it is a bad way to program a short delay. one microsecond delay makes the ISRs 3 times longer than they need to be. 
Another possibility to consider would be to change the direction pins in the main loop as soon as each motor speed is calculated. we would hope that the timer wouldn't trigger until at least 250ns later, but if it did, the direction change would just get caught one step late which would probably be fine.

andrewda, 2019-06-28:
I _realllllllly_ dislike the 72 NOPs. For one, what if we needed to increase to 80 NOPs for whatever reason. Do we have to count them all to make sure we get the right number? Seems pretty ridiculous compared to something like `for` loop...

joshua255, 2019-06-28:
I'd be happy work any of these solutions
Use a delayMicroseconds(1)
Move the direction pin writes into loop
While(i<num){i++;} with num being how many increments takes 200ns worth of cpu time
What solution do you like best?

andrewda, 2019-06-28:
> While(i<num){i++;} with num being how many increments takes 200ns worth of cpu time

Why `while` instead of `for`? Also @circuitbird any suggestions?

joshua255, 2019-06-28:
The compiler might optimize out empty loops.

andrewda, 2019-06-28:
I _think_ if you define `i` as `volatile` it'll check every time and not optimize it out, but could be wrong. Something like this:

```
for(volatile int i = 0; i < cycles; i++);
```

Which definitely isn't the ugliest line of code ever.

joshua255, 2019-06-28:
How many processor cycles is one for loop iteration?

andrewda, 2019-06-28:
That question is well above my C low-level programming knowledge, but I can't imagine how it'd be any more than a `while` loop.

andrewda, 2019-06-28:
I just tested this function locally and it worked (albeit the timing being different due to a different processor):

```
void delay(int timeNs)
{
  int cycles = timeNs * 0.24;  // converts from nanoseconds to clock cycles
  for(volatile int i = 0; i < cycles; i++);
}
```

It seemed to work the same with and without the `volatile` keyword.

joshua255, 2019-06-28:
It doesn't matter if it takes 20 cycles or 1, as long as it is consistent
and we know how many loops we need to do.

circuitbird, 2019-06-29:
I'm not familiar with the Xtensa's instruction set, and I really don't know where to find the code in the disassembly because freeRTOS is always included and it takes up ~190kb

joshua255, 2019-06-29:
@circuitbird what's your opinion of the nicest solution here? NOPS? delayMicrosecond? setting the direction pins in loop?

circuitbird, 2019-06-30:
I would just do NOPs in a loop since the timing doesn't need to be that precise and delayMicroseconds() may not work correctly in an ISR. Something like `for(byte itr = 0; itr < somevalue, itr++) NOP();`, perhaps.

closed 2019-08-02 by joshua255