SouthEugeneRoboticsTeam/ursa #27: Hard code address

andrewda, 2019-06-30:
It seems as though the default address is `192.168.4.1` but it'd be good to hard code that in. If we do get this working, it might be nice to set the address to `10.25.21.1` for no reason other than that's what the FRC robots use.

joshua255, 2019-06-30:
`  WiFi.softAPConfig(IPAddress(10, 25, 21, ROBOT_ID), IPAddress(10, 25, 21, ROBOT_ID), IPAddress(255, 255, 255, 0));`
seems to work! This change will be in my wifi-testing pull request after I check some more things.

joshua255, 2019-06-30:
Would you prefer the robot_id as the last number so each robot is identifiable if they were all connected together (but then softAPConfig wouldn't be used anyways) or do you want a 1 on the end as you said?

circuitbird, 2019-06-30:
I think it should be constant because the driver station would need to know the last digit if it changed on each robot, and that would add an unnecessary step to setting them up.

andrewda, 2019-06-30:
I agree with @circuitbird, though it might be nice to have a nice flag at the top of the file which can configure to make it the robot id instead.

joshua255, 2019-06-30:
What purpose would a flag to just change the IP address serve? An interesting enhancement to consider would be putting a flag in the code to make the robot switch to connecting to a wifi network instead of creating one.

circuitbird, 2019-06-30:
Perhaps changing ip/connecting to a network should be features for another branch since so many changes to the code will be required. It's totally possible to put both modes is the same file and have some kind of selector, but I think the resulting clutter would not be worth it.

andrewda, 2019-06-30:
Yeah either of those would work. A physical switch on the robot would be cool too.

andrewda, 2019-06-30:
Just for reference:

https://github.com/espressif/arduino-esp32/issues/985#issuecomment-359157428

joshua255, 2019-06-30:
the hardcoded address has been very reliable. The reference above suggested a delay, so I put the gyro zeroing code before setting the access point, because it causes a short delay.

closed 2019-06-30 by joshua255