SouthEugeneRoboticsTeam/ursa #28: measure wifi signal strength

joshua255, 2019-06-30:
anyone have a command that works? WiFi.RSSI always equals 0

andrewda, 2019-06-30:
I just realized that I don't think this is possible. The client can check it's WiFi strength but not the device hosting the AP (that will always be full strength, i.e. 0).

joshua255, 2019-06-30:
It wouldn't be impossible to measure the signal strength of the messages sent by the client, would it? That doesn't mean we can do it on the esp32. @jandan2 do you think the driver station could measure the signal strength? The operating system will display it to, and it's not too important of a feature.

andrewda, 2019-06-30:
I don't _think_ that's possible, especially since we're using UDP which literally doesn't care about the receiver in any way, shape or form. It just throws the packets onto the network and hopes they're received. Plus, signal strength isn't measured on a per-packet basis, so I think it really does need to happen from the controller.

andrewda, 2019-06-30:
Actually I might be wrong about that... But the likelihood that the ESP32 WiFi chip has the hardware/antennas required to pull this off are slim to nil.

joshua255, 2019-07-16:
I don't think this can happen on the robot end-only the driverstation end. We can reopen this issue if we somehow find a way that might work. also close #17

closed 2019-07-16 by joshua255