SouthEugeneRoboticsTeam/ursa #31: core panicked-debug amount

joshua255, 2019-07-01:
If I tell the compiler to use `Core Debug Level` of `None` the program crashes and I get these error messages.

```dhcps: send_offer>>udp_sendto result 0
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x00000000  PS      : 0x00060330  A0      : 0x800eff79  A1      : 0x3ffb40e0  
A2      : 0x3ffcd0e8  A3      : 0x3ffcd31c  A4      : 0x3ffcd0d0  A5      : 0x3ffcca08  
A6      : 0x0204a8c0  A7      : 0x6504a8c0  A8      : 0x800efe1c  A9      : 0x3ffb40a0  
A10     : 0x3ffcd0f8  A11     : 0x3ffcd31c  A12     : 0x3ffb40ec  A13     : 0x00000044  
A14     : 0x00000001  A15     : 0x00000006  SAR     : 0x00000010  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  

Backtrace: 0x00000000:0x3ffb40e0 0x400eff76:0x3ffb4120 0x400fb1e5:0x3ffb4140 0x400feb7d:0x3ffb4180 0x40103db2:0x3ffb41a0 0x400f2d63:0x3ffb41c0 0x4008a139:0x3ffb41f0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6400
entry 0x400806a8
dhcps: send_offer>>udp_sendto result 0
```

circuitbird, 2019-07-02:
Program counter is at zero?? That's no good.

circuitbird, 2019-07-02:
https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/fatal-errors.html
This page suggests that the cause of the error could be calling an invalid function pointer. Perhaps there is some other code expecting the debug functions to be present?

closed 2019-08-02 by joshua255