SouthEugeneRoboticsTeam/ursa #36: Split up code

circuitbird, 2019-07-07:
`ursa.ino` is getting a bit long. I think that it should split up into smaller files.

joshua255, 2019-07-07:
good point! I made a commit to the dev-tabs branch that puts wifi and gyro code in seperate tabs. I would like to split more things into tabs, but when I tried to move the stepper code I ran into problems with the order of declarations. How should we split the code but still make everything get declared in the right order? Also, should the variable declarations at the top be split up? I've never gotten .h and .cpp files to work for me, is that what's necessary?

joshua255, 2019-07-07:
please let me know what you think of the way I split the code up in pull request [#38](https://github.com/SouthEugeneRoboticsTeam/ursa/pull/38)

closed 2019-07-07 by joshua255