SouthEugeneRoboticsTeam/ursa #37: read battery voltage

joshua255, 2019-07-07:
Use analogRead from a pin that will be connected to a voltage divider to measure the robot's battery voltage. 0=0v 255=13v

andrewda, 2019-07-07:
in the driver station, should we display battery voltage as a percentage or as a voltage?

joshua255, 2019-07-07:
What do you think of showing voltage as a number, and an approximate percentage as a bar graph filling the battery?

joshua255, 2019-07-07:
A few notes about esp32 analog reads: 
Using wifi means we can only use ADC1-connected to pins 32 - 39 for analog read
Nice reference for analog read functions, and a note that the analog read function may not be linear, so we may have to calibrate with a voltmeter before getting accurate readings. [https://randomnerdtutorials.com/esp32-adc-analog-read-arduino-ide/](https://randomnerdtutorials.com/esp32-adc-analog-read-arduino-ide/)

joshua255, 2019-07-07:
[Excellent reference for voltage dividers](http://blogs.4j.lane.edu/tubman_a/files/2018/11/11-20-2018-Electric-circuits-day-2.pdf) 
[sparkfun voltage divider reference](https://learn.sparkfun.com/tutorials/voltage-dividers/all)

`VOut=VIn*R2/(R1+R2)

closed 2019-07-10 by joshua255