SouthEugeneRoboticsTeam/ursa #39: Save and recall pid settings

joshua255, 2019-07-08:


joshua255, 2019-07-08:
@andrewda any ideas for how the robot can know when to save its values?

joshua255, 2019-07-08:
Here's my ideas for what to do:

- [ ] make functions to save variables to memory that lasts between power cycles

- [ ] on startup, recall saved variables

- [ ] send saved settings to DS either until acknowledgement from DS or at request of the DS.

- [ ] save settings, only when necessary

circuitbird, 2019-07-09:
@joshua255 the [ESP32 EEPROM library](https://github.com/espressif/arduino-esp32/blob/master/libraries/EEPROM/src/EEPROM.cpp) already has functions for reading/writing all primitive data types. Also, some of the write functions were still writing to the wifi send buffer. I've switched the code to use the API functions in 918057ca266ca13b3cca50a7b7ebe225a72397e2.

joshua255, 2019-07-09:
Thanks @circuitbird! Writing to wifi explains why everything was broken for me. The write float functions look very nice.

closed 2019-07-10 by joshua255